arduino/Arduino #11824: Serial port performs a reset on the Arduino board, unnecessarily, if an error occurred

rtek1000, 2022-09-26:
Hello, I would like to register a small bug:

- When we are going to compile the sketch, at the end of an error, the serial port performs a reset on the Arduino board, unnecessarily.

matthijskooijman, 2022-09-26:
This might be related to #8146.

rtek1000, 2022-09-26:
> 

👍

per1234, 2022-09-27:
Hi @rtek1000. Which version of the Arduino IDE are you using?

If you haven't already, please give the latest nightly build of Arduino IDE 2.x a try and then let us know whether the problem still occurs. The download links are listed here:

https://www.arduino.cc/en/software#nightly-builds

rtek1000, 2022-09-27:
> Hi @rtek1000. Which version of the Arduino IDE are you using?
> 
> If you haven't already, please give the latest nightly build of Arduino IDE 2.x a try and then let us know whether the problem still occurs. The download links are listed here:
> 
> https://www.arduino.cc/en/software#nightly-builds

Hello,

I am using Arduino IDE 1.8.19,

I'll try that version 2.0, thanks.

(I've tried this 2.0 in the past, but still haven't had esp8266 OTA working via mDNS, I use [esp-link](https://github.com/jeelabs/esp-link)  as UNO Wifi)

per1234, 2022-10-11:
Hi @rtek1000. Did you ever get the chance to give Arduino IDE 2.x a try? If so, did the problem still occur using that version?

rtek1000, 2022-10-20:
Hi,

I tested the IDE 2.0 and this problem did not occur.

I noticed a long delay for the IDE 2.0 to show the "About" window. Is there any way to monitor the resources this IDE is using to find out if the IDE is worth using?

Apparently I don't see any advantage in not using the IDE 1.8.19. Furthermore, it still does not support upload via TCP/IP connection with automatic device search by mDNS, for ESP8266 and similar.

per1234, 2022-10-21:
>I tested the IDE 2.0 and this problem did not occur.

Excellent. Thanks so much for checking!

>I noticed a long delay for the IDE 2.0 to show the "About" window.

That issue is tracked here: https://github.com/arduino/arduino-ide/issues/1572

>Is there any way to monitor the resources this IDE is using to find out if the IDE is worth using?

Sure. You can do that just as you would with any application. If you would like guidance in doing so, please post on the Arduino forum and I'll be happy to help you out:

https://forum.arduino.cc/c/software/arduino-ide-2-0/93

But note this specific instance of the slow dialog opening is not really a matter of resource usage as we normally think of it. It is a matter of sending an HTTP request and then waiting for a response.

closed 2022-10-21 by per1234