arduino/Arduino #11835: Error, when try second compiling.

CodeNameHawk, 2022-11-04:
First compilation(verify) works normal.
Second compiling has error:
`C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Marian\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Marian\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Marian\Documents\Arduino\libraries -fqbn=STMicroelectronics:stm32:GenF1:pnum=BLUEPILL_F103C8,upload_method=swdMethod,xserial=generic,usb=none,xusb=FS,opt=osstd,dbg=none,rtlib=nano -vid-pid=0403_6001 -ide-version=10819 -build-path C:\Users\Marian\AppData\Local\Temp\arduino_build_546186 -warnings=none -build-cache C:\Users\Marian\AppData\Local\Temp\arduino_cache_69633 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-10.3.1-2.3.path=C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.STM32Tools.path=C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -prefs=runtime.tools.STM32Tools-2.1.1.path=C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -prefs=runtime.tools.CMSIS.path=C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -verbose C:\Users\Marian\Documents\Arduino\STM32\Podworz_Temp_STM32\Podworz_Temp_STM32.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Marian\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Marian\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Marian\Documents\Arduino\libraries -fqbn=STMicroelectronics:stm32:GenF1:pnum=BLUEPILL_F103C8,upload_method=swdMethod,xserial=generic,usb=none,xusb=FS,opt=osstd,dbg=none,rtlib=nano -vid-pid=0403_6001 -ide-version=10819 -build-path C:\Users\Marian\AppData\Local\Temp\arduino_build_546186 -warnings=none -build-cache C:\Users\Marian\AppData\Local\Temp\arduino_cache_69633 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-10.3.1-2.3.path=C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3 -prefs=runtime.tools.STM32Tools.path=C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -prefs=runtime.tools.STM32Tools-2.1.1.path=C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.1.1 -prefs=runtime.tools.CMSIS.path=C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -prefs=runtime.tools.CMSIS-5.7.0.path=C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\tools\CMSIS\5.7.0 -verbose C:\Users\Marian\Documents\Arduino\STM32\Podworz_Temp_STM32\Podworz_Temp_STM32.ino
Using board 'GenF1' from platform in folder: C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0
Using core 'arduino' from platform in folder: C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0
"C:\\Users\\Marian\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\STM32Tools\\2.1.1/win/busybox.exe" sh "C:\\Users\\Marian\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/extras/prebuild.sh" "C:\\Users\\Marian\\AppData\\Local\\Temp\\arduino_build_546186" "C:\\Users\\Marian\\Documents\\Arduino\\STM32\\Podworz_Temp_STM32" "C:\\Users\\Marian\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0"
Detecting libraries used...
Error compiling for board Generic STM32F1 series.`

Win 10
ArduinoIDE 1.8.19
Arduino Core for STM32 2.3.0
Board STM32F103C8T6

per1234, 2022-11-04:
Hi @CodeNameHawk. Thanks for your report.

Please try Arduino IDE 2.x to see whether the problem still occurs with that version of the IDE.

The download links are listed here:

https://www.arduino.cc/en/software/

CodeNameHawk, 2022-11-05:
In arduino 2.0.0, second compilation
`Using board 'GenF1' from platform in folder: C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0
Using core 'arduino' from platform in folder: C:\Users\Marian\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.3.0
"C:\\Users\\Marian\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\STM32Tools\\2.1.1/win/busybox.exe" sh "C:\\Users\\Marian\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/extras/prebuild.sh" "C:\\Users\\Marian\\AppData\\Local\\Temp\\arduino-sketch-A9B74F906728D4D8E8AC6FC519B02ED8" "C:\\Users\\Marian\\Documents\\Arduino\\STM32\\Podworz_Temp_STM32" "C:\\Users\\Marian\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0"
Detecting libraries used...
read ECONNRESET

Compilation error: read ECONNRESET`

CodeNameHawk, 2022-11-05:
VS Code
`[Starting] Verifying sketch 'Podworz_Temp_STM32.ino'
Please see the build logs in output path: c:\Users\Marian\Documents\Arduino\STM32\build
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
[Warning] Failed to generate IntelliSense configuration.
[Error] Verifying sketch 'Podworz_Temp_STM32.ino': Exit with code=1`

per1234, 2022-11-05:
Thanks!

I'm going to ask you to provide some additional information that might help us to identify the problem.

Please do this:

1. Start Windows "**File Explorer**".
1. Open the Arduino IDE 2.x installation folder
   (usually `C:\Program Files\Arduino IDE` or `C:\Users\<user name>\AppData\Local\Programs\Arduino IDE`).
1. In the folder listing of the Arduino IDE installation folder, hold the **<kbd>Shift</kbd>** key while clicking the right hand button on the mouse.
1. From the context menu, click "**Open PowerShell window here**".
   PowerShell will now open with the current directory set to the Arduino IDE installation folder.
1. Type the following command:
   ```text
   & ".\Arduino IDE"
   ```
1. Press the **<kbd>Enter</kbd>** key.
   Debug output should now be printed to the PowerShell window as the Arduino IDE starts (or fails to start).
1. Take whatever actions are required to trigger the compilation error, just as you did before.
1. Switch back to the PowerShell window.
1. Right click on the title bar of the PowerShell window.
   This will open a context menu.
1. Select **Edit > Select All** from the context menu.
   This will select all the text in the PowerShell window.
1. Press **<kbd>Ctrl</kbd>+<kbd>C</kbd>**.
   This will copy the contents of the PowerShell window to the clipboard.
1. Paste (**<kbd>Ctrl</kbd>+<kbd>V</kbd>**) the output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

CodeNameHawk, 2022-11-05:
log Start AdrduinoIde:
`PS C:\Program Files\Arduino IDE> & '.\Arduino IDE.exe'
PS C:\Program Files\Arduino IDE> [4448:1105/172046.585:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is disabled, ANGLE is nd process. PID: 8236
daemon INFO {"level":"info","msg":"from discovery builtin:mdns-discovery received message type: add, port: 192.168.1.26","time":"2022-11-05T17:20:50+02:00"}.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///c%3A/Users/Marian/.theia'
discovery-log INFO {"eventType":"add","port":{"matchingBoardsList":[],"port":{"address":"192.168.1.26","label":"Kotlownia at 192.168.1.26","protocol":"network","protocolLabel":"Network Port","propertiesMap":[[".","\\\"ESP8266_NODEMCU\\\""],["auth_upload","no"],["board","\\\"ESP8266_NODEMCU\\\""],["hostname","Kotlownia.local."],["port","8266"],["ssh_upload","no"],["tcp_check","no"]]}},"error":""}Contribution.configure: 0.5 ms [Finished 8.471 s after backend start]
discovery-log WARN Port 'Kotlownia at 192.168.1.26 Network Port' was already available. Known boards before override: []
root INFO Attached boards and available ports changed:igure: 0.3 ms [Finished 8.471 s after backend start]
root INFO No changes.alizationBackendContribution.initialize took longer than the expected maximum 50 milliseconds: 315.
root INFO ------------------------------------------
root INFO Backend TaskBackendApplicationContribution.onStart: 0.3 ms [Finished 8.471 s after backend start]
root INFO Backend ElectronTokenBackendContribution.configure: 2.5 ms [Finished 8.474 s after backend start]
root INFO Backend PluginDeployerContribution.initialize: 0.7 ms [Finished 8.474 s after backend start]
root INFO Backend ConfigServiceImpl.onStart: 1.2 ms [Finished 8.475 s after backend start]
root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.7 ms [Finished 8.477 s after backend start]
root INFO Backend ArduinoDaemonImpl.onStart: 1.9 ms [Finished 8.479 s after backend start]
root INFO Backend NodeFileUploadService.configure: 3.7 ms [Finished 8.479 s after backend start]
daemon INFO Starting daemon from C:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
discovery-log INFO start
discovery-log INFO start new deferred
root INFO Backend BoardDiscovery.onStart: 1.4 ms [Finished 8.481 s after backend start]
root INFO Backend FileDownloadEndpoint.configure: 1.0 ms [Finished 8.482 s after backend start]
root INFO Backend AuthenticationServiceImpl.onStart: 0.5 ms [Finished 8.483 s after backend start]
root INFO Backend PluginApiContribution.configure: 0.7 ms [Finished 8.484 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.6 ms [Finished 8.485 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.1 ms [Finished 8.486 s after backend start]
root INFO Backend PlotterBackendContribution.configure: 9.2 ms [Finished 8.495 s after backend start]
root INFO Theia app listening on http://localhost:49774.
root INFO Finished starting backend application: 18.8 ms [Finished 8.503 s after backend start]
root WARN The local plugin referenced by local-dir:/c%3A/Users/Marian/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/Marian/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:C:\Users\Marian\.arduinoProIDE\plugins does not exist.
Restoring workspace roots: c:\Users\Marian\Documents\Arduino\STM32\Podworz_Temp_STM32
isTempSketch: false. Input was c:\Users\Marian\Documents\Arduino\STM32\Podworz_Temp_STM32
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.5" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-cs" to a VS Code extension "vscode-language-pack-cs@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-de" to a VS Code extension "vscode-language-pack-de@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-es" to a VS Code extension "vscode-language-pack-es@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-fr" to a VS Code extension "vscode-language-pack-fr@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-it" to a VS Code extension "vscode-language-pack-it@1.69.0" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
[... cut ...]
daemon INFO {"level":"info","msg":"from discovery builtin:serial-discovery received message type: hello, message: OK, protocol version: 1","time":"2022-11-05T17:20:05+02:00"}
{"level":"info","msg":"sending command START_SYNC to discovery builtin:serial-discovery","time":"2022-11-05T17:20:05+02:00"}
{"level":"info","msg":"from discovery builtin:serial-discovery received message type: start_sync, message: OK","time":"2022-11-05T17:20:05+02:00"}`

Press Verify and get error:
[image]
The log has not been updated.

CodeNameHawk, 2022-11-05:
Additional information may be helpful.
If I try to uninstall ArduinoIDE 2.0.0 I get an error
[image]

Only one Arduino process is visible in running processes.
If I close the uninstaller, there are no processes named arduino.

per1234, 2022-11-05:
>If I try to uninstall ArduinoIDE 2.0.0 I get an error

This bug is being tracked at https://github.com/arduino/arduino-ide/issues/1163

CodeNameHawk, 2022-11-06:
> > If I try to uninstall ArduinoIDE 2.0.0 I get an error
> 
> This bug is being tracked at [arduino/arduino-ide#1163](https://github.com/arduino/arduino-ide/issues/1163)

Removing arduinoide 2.0.0 and arduinoide 1.8.19 with the Advanced_Uninstaller13 and installing a new arduinoide 2.0.1 did not solve the compilation problem.

ps. arduinoide 2.0.1 is not removed by means of win10, only with the help of Advanced_Uninstaller13.

CodeNameHawk, 2022-11-07:
Did an experiment:
Uninstalled arduinoide 1.8.19 and arduino ide 2.0.1, remove folder with sketch and folder all packages.
After reboot installed arduino ide 2.0.1, selected ArduinoUno board and compile.
The error happened again.
`Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS C:\Windows\system32> cd..
PS C:\Windows> cd..
PS C:\> cd '.\Program Files\'
PS C:\Program Files> cd "C:\Users\Marian\AppData\Local\Programs\Arduino IDE\"
PS C:\Users\Marian\AppData\Local\Programs\Arduino IDE> & '.\Arduino IDE.exe'
PS C:\Users\Marian\AppData\Local\Programs\Arduino IDE>
Arduino IDE 2.0.1
Starting backend process. PID: 6160
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///c%3A/Users/Marian/.theia'
root INFO Backend Object.initialize: 6.8 ms [Finished 1.646 s after backend start]
root INFO Backend Object.configure: 5.8 ms [Finished 1.646 s after backend start]
root INFO Backend MessagingContribution.onStart: 3.4 ms [Finished 1.646 s after backend start]
root INFO Backend OSBackendApplicationContribution.configure: 0.5 ms [Finished 1.690 s after backend start]
root INFO Backend DefaultWorkspaceServer.onStart: 0.4 ms [Finished 1.690 s after backend start]
root INFO Backend LocalizationBackendContribution.configure: 0.2 ms [Finished 1.691 s after backend start]
root INFO Backend LocalizationBackendContribution.initialize: 44.0 ms [Finished 1.691 s after backend start]
root INFO Backend TaskBackendApplicationContribution.onStart: 0.3 ms [Finished 1.691 s after backend start]
root INFO Backend ElectronTokenBackendContribution.configure: 2.3 ms [Finished 1.694 s after backend start]
root INFO Backend PluginDeployerContribution.initialize: 0.5 ms [Finished 1.694 s after backend start]
root INFO Backend ConfigServiceImpl.onStart: 1.0 ms [Finished 1.695 s after backend start]
root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.4 ms [Finished 1.696 s after backend start]
root INFO Backend ArduinoDaemonImpl.onStart: 1.3 ms [Finished 1.697 s after backend start]
root INFO Backend NodeFileUploadService.configure: 2.6 ms [Finished 1.697 s after backend start]
daemon INFO Starting daemon from C:\Users\Marian\AppData\Local\Programs\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
discovery-log INFO start
discovery-log INFO start new deferred
root INFO Backend BoardDiscovery.onStart: 1.0 ms [Finished 1.699 s after backend start]
root INFO Backend FileDownloadEndpoint.configure: 0.6 ms [Finished 1.699 s after backend start]
root INFO Backend AuthenticationServiceImpl.onStart: 0.5 ms [Finished 1.701 s after backend start]
root INFO Backend PluginApiContribution.configure: 0.6 ms [Finished 1.701 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.5 ms [Finished 1.702 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.1 ms [Finished 1.703 s after backend start]
root INFO Backend PlotterBackendContribution.configure: 1.1 ms [Finished 1.704 s after backend start]
root INFO Theia app listening on http://localhost:62894.
root INFO Finished starting backend application: 6.0 ms [Finished 1.708 s after backend start]
root WARN The local plugin referenced by local-dir:/c%3A/Users/Marian/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/Marian/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:C:\Users\Marian\.arduinoProIDE\plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.5" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-bg" to a VS Code extension "vscode-language-pack-bg@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-cs" to a VS Code extension "vscode-language-pack-cs@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-de" to a VS Code extension "vscode-language-pack-de@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-es" to a VS Code extension "vscode-language-pack-es@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-fr" to a VS Code extension "vscode-language-pack-fr@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
[... cut ...]
INFO[0064] Adding libraries dir                          dir="C:\\Users\\Marian\\Documents\\Arduino\\libraries" location=user

discovery-log INFO error received
discovery-log ERROR Unexpected error occurred during the boards discovery.
discovery-log INFO received end
root ERROR Have not received the build path from the CLI while running the compilation.
daemon INFO Daemon exited with exit code: 3221226505.
[3524:1107/104727.805:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is disabled, ANGLE is
                                                     >`

per1234, 2022-11-07:
There are some strange characters in the response from the "discovery" tools that find the ports of Arduino boards connected to your computer:

```text
INFO[0001] from discovery builtin:serial-discovery received message type: hello, ╤Б╨╛╨╛╨▒╤Й╨╡╨╜╨╕╨╡: OK, ╨▓╨╡╤А╤Б╨╕╤П ╨┐╤А╨╛╤В╨╛╨║╨╛╨╗╨░: 1
INFO[0001] sending command START_SYNC to discovery builtin:serial-discovery
INFO[0001] from discovery builtin:mdns-discovery received message type: hello, ╤Б╨╛╨╛╨▒╤Й╨╡╨╜╨╕╨╡: OK, ╨▓╨╡╤А╤Б╨╕╤П ╨┐╤А╨╛╤В╨╛╨║╨╛╨╗╨░: 1
INFO[0001] sending command START_SYNC to discovery builtin:mdns-discovery
INFO[0001] from discovery builtin:serial-discovery received message type: start_sync, ╤Б╨╛╨╛╨▒╤Й╨╡╨╜╨╕╨╡: OK
INFO[0001] from discovery builtin:mdns-discovery received message type: start_sync, ╤Б╨╛╨╛╨▒╤Й╨╡╨╜╨╕╨╡: OK

[...]

daemon INFO INFO[0001] from discovery builtin:mdns-discovery received message type: add, ╨┐╨╛╤А╤В: 192.168.1.26
```

Do you have any idea what might cause this?

Do you see those same characters when you look at those lines in the PowerShell window?

CodeNameHawk, 2022-11-07:
I have an additional RU language installed, but I don’t know why these characters are visible here.
Under the address 192.168.1.26, an OTA board works. 
[image]

CodeNameHawk, 2022-11-07:
If change Regional locale to United States
and see normal symbols in log:
`PS C:\Windows\system32> cd "C:\ProgramData\Arduino IDE"
PS C:\ProgramData\Arduino IDE> & '.\Arduino IDE.exe'
PS C:\ProgramData\Arduino IDE>
Arduino IDE 2.0.1
Starting backend process. PID: 1684
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///c%3A/Users/Marian/.theia'
root INFO Backend Object.initialize: 5.7 ms [Finished 1.531 s after backend start]
root INFO Backend Object.configure: 4.7 ms [Finished 1.531 s after backend start]
root INFO Backend MessagingContribution.onStart: 2.5 ms [Finished 1.531 s after backend start]
root INFO Backend OSBackendApplicationContribution.configure: 0.5 ms [Finished 1.575 s after backend start]
root INFO Backend DefaultWorkspaceServer.onStart: 0.4 ms [Finished 1.575 s after backend start]
root INFO Backend LocalizationBackendContribution.configure: 0.2 ms [Finished 1.576 s after backend start]
root INFO Backend LocalizationBackendContribution.initialize: 44.7 ms [Finished 1.576 s after backend start]
root INFO Backend TaskBackendApplicationContribution.onStart: 0.3 ms [Finished 1.576 s after backend start]
root INFO Backend ElectronTokenBackendContribution.configure: 1.9 ms [Finished 1.578 s after backend start]
root INFO Backend PluginDeployerContribution.initialize: 0.4 ms [Finished 1.578 s after backend start]
root INFO Backend ConfigServiceImpl.onStart: 0.9 ms [Finished 1.579 s after backend start]
root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.4 ms [Finished 1.580 s after backend start]
root INFO Backend ArduinoDaemonImpl.onStart: 1.3 ms [Finished 1.581 s after backend start]
root INFO Backend NodeFileUploadService.configure: 2.5 ms [Finished 1.581 s after backend start]
daemon INFO Starting daemon from C:\ProgramData\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
discovery-log INFO start
discovery-log INFO start new deferred
root INFO Backend BoardDiscovery.onStart: 1.1 ms [Finished 1.583 s after backend start]
root INFO Backend FileDownloadEndpoint.configure: 0.7 ms [Finished 1.584 s after backend start]
root INFO Backend AuthenticationServiceImpl.onStart: 0.4 ms [Finished 1.585 s after backend start]
root INFO Backend PluginApiContribution.configure: 0.6 ms [Finished 1.585 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.5 ms [Finished 1.587 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.1 ms [Finished 1.587 s after backend start]
root INFO Backend PlotterBackendContribution.configure: 1.0 ms [Finished 1.588 s after backend start]
root INFO Theia app listening on http://localhost:55925.
root INFO Finished starting backend application: 7.4 ms [Finished 1.593 s after backend start]
root WARN The local plugin referenced by local-dir:/c%3A/Users/Marian/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/Marian/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:C:\Users\Marian\.arduinoProIDE\plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.5" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-bg" to a VS Code extension "vscode-language-pack-bg@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-cs" to a VS Code extension "vscode-language-pack-cs@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-de" to a VS Code extension "vscode-language-pack-de@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-es" to a VS Code extension "vscode-language-pack-es@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-fr" to a VS Code extension "vscode-language-pack-fr@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-hu" to a VS Code extension "vscode-language-pack-hu@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-it" to a VS Code extension "vscode-language-pack-it@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-ja" to a VS Code extension "vscode-language-pack-ja@1.53.2" with engines:
[... cut ...] \\"ESP8266_NODEMCU\\\""],["hostname","Kotlownia.local."],["port","8266"],["ssh_upload","no"],["tcp_check","no"]]}},"error":""}
root INFO Attached boards and available ports changed:
root INFO  - New port is available on Kotlownia at 192.168.1.26 Network Port
root INFO ------------------------------------------
root ERROR Have not received the build path from the CLI while running the compilation.`

ps. i reinstalled arduino to new path: C:\ProgramData\Arduino IDE
But compilation error stil here.

CodeNameHawk, 2022-11-10:
If, in VS CODE, the "output" option is removed from the "arduino.json" file, the compilation starts working normally.

Most likely, arduino is locking the files of the previous compilation.

CodeNameHawk, 2022-12-19:
I found the cause of the error.
I need to delete the entry in the register, as described here: https://www.advanceduninstaller.com/Arduino-IDE-2_0_0-beta_11-7d455f8a9f283d5d088f74a25fe09760-application.htm

        HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Uninstall\80a6dee0-faee-5422-9648-6b7ee9b05f5a

I have numbers at the end : 459fc68c-eb53-59f8-8957-9913bc627af3

closed 2022-12-20 by CodeNameHawk