arduino/Arduino #11842: Updating/Downgrading Arduino-Pico core fails at file extraction

maxgerhardt, 2022-11-27:
## Versions
Arduino IDE: Latest 1.8.19
OS: Windows 11, (`ver`: `Version 10.0.22000.1219`), in Oracle VirtualBox from [Microsoft](https://developer.microsoft.com/de-de/windows/downloads/virtual-machines/)
[Arduino-Pico](https://github.com/earlephilhower/arduino-pico): 2.6.3 -> 2.6.1

## Description

Per https://github.com/earlephilhower/arduino-pico/issues/976.

Users report that when upgrading or downgrading the Arduino-Pico core from 2.6.1 to 2.6.3 (or reverse), updating of the core would fail at the Python interpreter. This leaves the Python interpreter package folder without some crucial files without which it cannot start, and so users get a 
```
Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'
```
when attempting to further compile something.

## Reproduction

The steps I used (some may not be needed though):
1. Install the Arduino IDE from the .zip package for Windows
2. Install Arduino-Pico's latest version [as described](https://github.com/earlephilhower/arduino-pico#installing-via-arduino-boards-manager)
3. Compile a blink sketch for the Pico. It should compile normally
4. Close the Arduino IDE
5. Install Python3 from https://www.python.org/downloads/, tick the checkmark for "Add python to PATH"
6. Reopen the Arduino IDE
7. Go to the board manager to downgrade the RP2040 core to 2.6.1

## Expected Behavior

The downgrade is done successfully.

## Actual behavior

The downgrade fails with

[image]

The Python folder is half-deleted / upgraded, with many files missing, crippling it so heavily that `python3.exe` cannot start anymore.

[image]

Should look like 

[image]

maxgerhardt, 2022-12-12:
More users are running into this problem [here](https://github.com/earlephilhower/arduino-pico/issues/1039). Any insights?

per1234, 2022-12-14:
@maxgerhardt please check whether the problem occurs with the latest version of Arduino IDE 2.x and then comment here to let me know the result.

NicolaP8, 2023-01-06:
Hi,
I got the same error multiple times with IDE 1.8.19 and versions 2.6.x/2.7.0.
I had to remove the board then reinstall.

I've Win 10 with Python 3.11 installed and added to PATH.

AeroSynth, 2023-06-21:
I get the same error with 3.3.0
However, if I remove it and use 3.2.2, it compiles.
Setup:  
Windows 10
Python 3.9.5
Arduino IDE 1.8.19

So something is broken with 3.3.0