arduino/Arduino #11851: Upload with Arduino IDE 2.0.3 fails.

plunder2, 2022-12-26:
Copy from output window:
"/Users/heinrichmueller/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8/bin/avrdude" "-C/Users/heinrichmueller/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8/etc/avrdude.conf" -v -V -patmega2560 -cwiring "-P/dev/cu.usbmodem1411301" -b115200 -D "-Uflash:w:/private/var/folders/02/2m5xmk0j5bb19k5jt4lnh46h0000gq/T/arduino-sketch-B44509711192F6CF6691A744BF3E1B30/Lueftung.ino.hex:i"
Failed uploading: cannot execute upload tool: fork/exec /Users/heinrichmueller/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8/bin/avrdude: bad CPU type in executable

Workaround is:
Open a terminal Window and paste the command from above but replace 6.3.0-arduino8 with 6.3.0-arduino17
and the hex file uploads just fine.

Mac book air 2018, Intel, Ventura 13.01
Target: Controllino MAXI Automation 

Regards
                  Heico

per1234, 2022-12-29:
Hi @plunder2. Thanks for taking the time to submit an issue report.

The 3rd party boards platform of the "Controllino MAXI Automation" board is using a very outdated version of the AVRDUDE tool. At the time that tool was built, all versions of macOS supported 32-bit binaries. Since that time, 32-bit support was dropped and binaries must be built for 64-bit in order to be used with modern versions of macOS. As you found, the recent versions of AVRDUDE are built accordingly.

So this is a problem with the 3rd party boards platform you are using and thus off topic for this repository. You should contact the maintainers of the boards platform about this problem.

closed 2022-12-29 by per1234

plunder2, 2022-12-29:
Thank you for take in the time to read my issue! I will try to connect the board maintainers. Regards Heico