arduino/Arduino #11856: Wrong shortcut for zoom in/out in macOS v1.8.16 with AZERTY keyboard

joyel24, 2023-01-24:
The shortcut to change the text size do not correspond to the keyboard mapping on macOS when keyboard is AZERTY.

per1234, 2023-01-24:
Hi @joyel24. Thanks for taking the time to submit an issue.

I'll assume the "v1.8.16" in your issue title is referring to the Arduino IDE version. If so, you are using a very outdated version of Arduino IDE. We do not support the old IDE versions. Please try Arduino IDE 2.x. The download links are listed here:

https://www.arduino.cc/en/software

If you find any problems with the modern version of Arduino IDE, you can report them to the appropriate issue tracker:

https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/issues.md#issue-report-guide

closed 2023-01-24 by per1234

joyel24, 2023-01-24:
Thanks @per1234 for your fast answer, I use the old IDE because I  can't upload my code via network with esp32 modules due to this issue : https://github.com/arduino/arduino-ide/issues/1488
Its asking a password but I don't want to set one so I'm forced to use the v1 IDE...

per1234, 2023-01-24:
>I can't upload my code via network with esp32 modules due to this issue : https://github.com/arduino/arduino-ide/issues/1488

Even though it is a poor user experience. That issue does not prevent you from uploading to network ports.

Just use this workaround:

1. Add any arbitrary character(s) to the optional "**Password**" field.
1. Click the "**UPLOAD**" button.

joyel24, 2023-01-24:
> Even though it is a poor user experience. That issue does not prevent you from uploading to network ports.
>
> Just use this workaround:
>
> 1. Add any arbitrary character(s) to the optional [...]

This is not a perfect fix but it allow to upload my sketch via network. Thank you 👍