arduino/Arduino #11858: Issue with some libraries with misleading claims

ssilverman, 2023-01-28:
I contributed an Ethernet library for the Teensy 4.1 called QNEthernet. I have an issue where a bunch of repos claim in their description to use it, however they do not. Instead, they use some underlying (and publicly-accessible) components which are specifically tailored to be used a different way than are actually used in those repos.

This has led to user confusion and also lots of noise in any search results.

I thought I’d ask here to see what my options are. I’ve already contacted the author about this, but my attempts at communication via repo issues were ignored and deleted. I’m happy to engage in a private discussion if that’s a better forum for this help request.

per1234, 2023-01-28:
Hi @ssilverman. Thanks for taking the time to submit an issue.

It sounds like an interesting subject to explore, but unless I misunderstand it doesn't seem to be directly related to the software or documentation content associated with this issue tracker. For this reason, I think it will be best to discuss the subject on the Arduino Forum instead of here. There is a dedicated category just for libraries:

https://forum.arduino.cc/c/development/libraries/39

I'll keep an eye out for your post and engage with you over there to see if I can make any suggestions once I have a better understanding of the situation.

If some ideas for how Arduino's software or documentation might be enhanced to mitigate such problems result from that discussion, please do submit formal proposals for the specific changes you would like to see Arduino make to their projects.

closed 2023-01-28 by per1234

ssilverman, 2023-01-30:
Thanks for the direction. Here is the new post:
https://forum.arduino.cc/t/issue-with-some-libraries-with-misleading-claims-about-use-of-another-library/1083933