arduino/Arduino #11863: Portenta H7 Issue

Bilalgp, 2023-02-13:
I am having an issue having pin A4 read and pin D10 digital output simultaneously. They appear to be connected in some way. If I run the following code, D10's output depends on the incoming analog signal. 

I tried multiple Portentas and all are suffering from the same problem. I am wondering if this is a hardware or software issue. I am integrating this portenta on a fixed circuit board, so changing the utilization of the pins is difficult. 

`int max_A0;
int max_A1;
int max_A2;

int sensorValue1, sensorValue2;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);

  //pinMode(15, INPUT);
  //pinMode(16, INPUT);
  //pinMode(18, INPUT);
  //pinMode(19, INPUT);
  //pinMode(20, INPUT);
  pinMode(10, OUTPUT);
  pinMode(0, OUTPUT);
  pinMode(4, OUTPUT);
  delay(1000);
  //digitalWrite(D10, OUTPUT);  // sets the digital power ON (D10)
  //digitalWrite(clock1, OUTPUT);  // sets the digital power ON (D10)
  digitalWrite(0, HIGH);  // sets the digital pin D4 off
  //digitalWrite(10, HIGH);  // sets the digital power ON (D10)
  //digitalWrite(4, HIGH);
  delay(1000);
  digitalWrite(10, HIGH);
}

int i = 0;
void loop() {
  // put your main code here, to run repeatedly:

  //digitalWrite(10, HIGH);
  delay(10);
  //digitalWrite(10, HIGH);
  //pinMode(19, INPUT);
  //delay(10000);
  //pinMode(19, INPUT);
  max_A0 = analogRead(20);
  delay(10);
  sensorValue1 = analogRead(15);
  delay(10);
  sensorValue2 = analogRead(16);
  delay(10);

  max_A2 = analogRead(18);
  delay(10);  
  //pinMode(19, INPUT);
  max_A1 = analogRead(A4);
  delay(10);
  //pinMode(10,OUTPUT);
  Serial.println(max_A1);
  //delay(1000);

  //digitalWrite(10, HIGH);

  //if(i < 5) {
  //  pinMode(10,OUTPUT);

  //pinMode(10,INPUT);
  // }
  i++;
  //int max_A1 = analogRead(19);
  // Serial.println(max_A1);
}`

Sincerely,

Bilal H.

per1234, 2023-02-15:
Hi @Bilalgp. Thanks for your report. It seems there is a lot of unnecessary cruft in your demonstration sketch that will muddy the subject. Please reduce it down to the absolute minimum amount of code necessary to reproduce the problem.

Does the problem occur if you upload this sketch to your Portenta board?:

```cpp
void setup() {
  pinMode(10, OUTPUT);
}
void loop() {}
```

Bilalgp, 2023-02-15:
Yes, this works. Once I add max_A1 = analogRead(A4); in the loop, the digital pin 10 drops to 0.

per1234, 2023-02-15:
Please provide the minimal demonstration sketch and detailed instructions we can follow to reproduce the problem.

Please also answer the following questions:

- What were the results you expected from following those instructions?
- What were the results you observed that did not match your expectations?

Bilalgp, 2023-02-15:
With all due respect, I don't think there is cruft in the code. Here is the updated code. 

   > What were the results you expected from following those instructions?
   

From here, we should expect D10 to remain high while simultaneously measuring the analog input A4. 

> What were the results you observed that did not match your expectations?

However, when measuring with an oscilloscope, D10 is not high anymore. It appears to match the value of A4. 

`
int max_A1;

void setup() {
  Serial.begin(9600);

  pinMode(10, OUTPUT);
  delay(1000);
  digitalWrite(10, HIGH);
}

void loop() {

  max_A1 = analogRead(A4);
  delay(10);
  Serial.println(max_A1);

}`