arduino/Arduino #11866: Error while uploading to esp32s3

Sirius3615, 2023-03-02:
Arduino: 1.8.19 (Mac OS X), Board: "ESP32S3 Dev Module, Disabled, Disabled, QIO 80MHz, 8MB (64Mb), Core 1, Core 1, Hardware CDC and JTAG, Disabled, Disabled, Disabled, UART0 / Hardware CDC, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), 921600, None, Disabled"

```
Sketch uses 1034761 bytes (78%) of program storage space. Maximum is 1310720 bytes.
Global variables use 48324 bytes (14%) of dynamic memory, leaving 279356 bytes for local variables. Maximum is 327680 bytes.
java.io.IOException: Cannot run program "___REMOVE___/esptool": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2061)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch

```

per1234, 2023-03-02:
Hi @Sirius3615. Thanks for your report. It seems the latest version of the 3rd party **esptool** tool is being detected as malicious by some security software during the installation or update of the ESP32 boards platform: https://github.com/espressif/esptool/issues/849.

When the file is put in quarantine by the security software it causes the error you encountered.

Please check your antivirus or other security software to see if it has quarantined the file, then comment here to let me know what you discovered.

per1234, 2023-03-03:
Hi @Sirius3615. I did some investigation and found the cause of the error was a change made in the ESP32 platform repository. I submitted a report there:

https://github.com/espressif/arduino-esp32/issues/7923

You can restore the platform to a functional state by uninstalling and then reinstalling it:

1. Select **Tools > Board > Boards Manager** from the Arduino IDE menus.
   A "**Boards Manager**" dialog will appear.
1. Wait for the updates to finish, as shown by the messages printed at the bottom of the "**Boards Manager**" dialog.
1. Scroll down through the list of boards platforms until you find the "**esp32 by Espressif Systems**" entry. Click on it.
   Some buttons will appear on the entry.
1. Click the **<kbd>Remove</kbd>** button on the "**esp32 by Espressif Systems**" entry.
1. Wait for the removal to finish.
1. Click the **<kbd>Install</kbd>** button on the "**esp32 by Espressif Systems**" entry.
1. Wait for the installation to finish.
1. Click the **<kbd>Close</kbd>** button on the "**Boards Manager**" dialog.

If you have any questions or problems while following those instructions, please post on the Arduino Forum. We'll be able to help you out over there:

https://forum.arduino.cc/

closed 2023-03-03 by per1234

Sirius3615, 2023-03-03:
Hi @per1234,
I am sorry for such a late response, I did everything as you said and **It works**. Thank you for taking your time to help me with this!