arduino/Arduino #11871: Keyboard library mentions Arduino Uno/Mega/Nano in the compatibility list

sterretjeToo, 2023-03-16:
https://www.arduino.cc/reference/en/libraries/keyboard/

Subject says it all ;)

matthijskooijman, 2023-03-16:
I suspect this list is autogenerated from the "architecture = avr" or so in the library.properties. I don't think libraries can explicitly state compatibility with some boars, but not others, so this should probably be fixed in the description instead (and looking at that page, the "compatibility note" even already does that, though it might be a bit more specific maybe).

per1234, 2023-03-16:
Hi @sterretjeToo.

@matthijskooijman already answered while I was writing, but I'll post anyway in case there is a little bit of additional information:

The library reference pages like https://www.arduino.cc/reference/en/libraries/keyboard/ are automatically generated for all the libraries in the Arduino Library Manager based on the information in their [`library.properties` metadata file](https://arduino.github.io/arduino-cli/latest/library-specification/#library-metadata).

The "Compatibility" list on the reference pages is based on the `architectures` field of that file. The reference page generator simply adds all the official boards of each architecture in the field to the list.

As in this case, a library might require capabilities that are not provided by all boards of that architecture. There is no machine readable data that would allow the generator to understand this so boards that are incompatible with the library may be listed in the reference page.

There has been some internal discussion about how the library metadata system might be expanded to indicate board compatibility, but for now we settled on warning the reader about the situation via the "Compatibility note" on the reference page:

> ### Compatibility Note
>
>Note: while the library is supposed to compile correctly on these architectures, it might require specific hardware features that may be available only on some boards.

sterretjeToo, 2023-03-16:
Unfortunately that note does not reveal anything to the beginner that consults the page. Please explain how somebody is to know what that specific hardware is?

Recent topic on the forum that triggered this issue: https://forum.arduino.cc/t/arduino-keyboard-not-working-on-arduino-uno-r3/1102400.

per1234, 2024-12-19:
> the "Compatibility note" on the reference page:
>
>> ### Compatibility Note
>>
>>Note: while the library is supposed to compile correctly on these architectures, it might require specific hardware features that may be available only on some boards.

The note has since been changed to one that is even more vague and cryptic, as well as being easier to overlook. It now says:

> Compatibility with an architecture means that code can be compiled and uploaded to a board from the list below:

The note is inaccurate since the list can contain boards for which the library will not compile. For example, it claims that the Keyboard library can be compiled for the UNO R3, classic Nano, and Mega boards:

https://docs.arduino.cc/libraries/keyboard/

per1234, 2025-09-04:
The reference page content is also potentially problematic in the case of libraries that use the `*` wildcard value, and a different manner than the others mentioned here since no caveat at all is displayed in this case. It only says:

> This library is compatible with **all** architectures so you should be able to use it on all the Arduino boards.

For example (though admittedly not the best one since there is a good chance this high level library  actually is compatible with all board):

https://docs.arduino.cc/libraries/stepper/#Compatibility