arduino/Arduino #11882: Auto format (arduino formatter) breaks this line in #define

littlegreenrock, 2023-05-04:
1st: apologies, as I am too novice to be able to identify what this is called.  Because of the esoteric nature of this line of code, this issue couldn't possibly be affecting ... well... maybe anyone more than myself.  Further, anyone who would use such a line would know how to work around it.  The question is more:  "how does this line of code get interpreted, and what does it mean?" rather than "can you fix this bug?"

**what should happen?** Autoformat shouldn't remove the 'space', or move the asterisk.
**Why is it doing this?** it looks a lot like a reference asterisk, probably the compiler can't tell the difference. (i can't)
**Repeatable?**  Yes, easily.
**Workaround?** Yes, with a tiny block quote
**Impact and Urgency?**  Low, Low. 

Example 1: here the line works as the block comment inhibits the interpreter from moving the asterisk.
```
#define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper /*dumb*/ *>(pstr_pointer))
```

Example 2a: Here the line is entered as to compile, working, 
```
#define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
```

Example 2b: ... after auto format the line is broken, will no longer compile.
```
#define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper*>(pstr_pointer))
```

per1234, 2023-05-04:
Hi @littlegreenrock. Thanks for your report.

Which version of Arduino IDE are you using?

littlegreenrock, 2023-05-23:
Arduino Version: 2.1.0
CLI Version: 0.32.2
Windows 10, 22H2 , build: 19045.2728