arduino/Arduino #11916: Project

dw1rm, 2023-10-17:


dw1rm, 2023-10-17:
[image]

dw1rm, 2023-10-17:
[image]

dw1rm, 2023-10-17:
[image]

The Digispark Attiny 85 design (schematic, layout, concept) is licensed under the Creative Commons Attribution-ShareAlike 3.0 Unported License: http://creativecommons.org/licenses/by-sa/3.0/
The Digispark and Digistump names and logos (or derivatives thereof) may not be used as part of the name of a product, company, or domain name without express written permission.
Digispark name. logo and Digistump name, logo Copyright 2013 Digistump LLC All Rights Reserved
For full details and license information please see http://digistump.com/wiki/digispark/policy

per1234, 2023-10-17:
Hi . Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This is not a "place to share experiences".

If you would like to discuss Arduino-related subjects, please use the Arduino Forum:

https://forum.arduino.cc/

closed 2023-10-17 by per1234