arduino/Arduino #11924: Adding Project Setup to the Projects by the IDE

purehunter1, 2023-11-05:
I use several different Arduino and ESP boards. It is impossible for me to memorize all the different setups for all the projects, like memory partitioning, core frequency, etc. .
Currently I use a workaround: writing all the setup parameters in a comment line in the source code. When opening an existing project, I have to setup these parameters in the IDE manually which is prone to errors and simply annoying.
With the increasing number of different boards, the situation gets messier every day.

So it would be helpful if the project setup would be stored with the source code file, e.g. in an additional file with the same filename as the source code, but a different filename extention.
When opening a source code file (.ino), the setup-file would be opened as well (by the IDE) and the stored setup being used as a preference. If the setup-file is missing, the behaviour of the IDE remains like today.

rgds

per1234, 2023-11-05:
Hi @purehunter1. Thanks for your suggestion. Arduino IDE 2.x already has an equivalent behavior, where the board selection, custom board options, and port selection that was used the last time you had a given sketch open are automatically selected when you open that sketch in Arduino IDE.

Arduino IDE 1.x is no longer under active development so the requested feature will not be added to Arduino IDE 1.x. The recommendation is for Arduino IDE 1.x users to update to 2.x to get access to the feature.

closed 2023-11-05 by per1234

purehunter1, 2023-11-05:
Hi per1234,
you are right. I did not notice it, as I also switch between the IDE's as the 2.x compiling is very slow compared with 1.8