arduino/Arduino #11945: Wayland Support

Master811129, 2024-02-07:
XWayland has many major issues on NVIDIA graphics cards: occasional flashing which is risky for people with epileptic issues, blurriness, etc.. adding support for native Wayland would be awesome.

per1234, 2024-02-07:
Hi @Master811129. Thanks for your interest in this open source project.

This is the issue tracker for Arduino IDE 1.x, which is no longer under active development. Arduino will not be making any efforts to resolve any lack of support for running Arduino IDE 1.x under native Wayland. If this is something you would like to do then you are welcome to fork this free open source software.

As for Arduino IDE 2.x, there is an existing issue tracking that deficiency: https://github.com/arduino/arduino-ide/issues/2107

closed 2024-02-07 by per1234

Master811129, 2024-12-16:
## For people who still have this issue

Nvidia added support for explicit sync and finally fixed the flashing issue for XWayland apps. For fixing the problem you need to switch to a more recent distro and use 560-or-higher driver series.

Good luck.