arduino/Arduino #12027: Bug Report: Random Parsing Errors in platform.txt (Linux Mint)

AdamKirby07, 2025-01-05:
Description:
On Linux Mint, I am encountering an issue where the Arduino IDE reports the following error when loading a locally created platform:

arduino
Copy code
loading platform release: Error reading file: Error parsing data at line 1: Invalid line format, should be 'key=value'
The platform.txt file is formatted correctly using the key=value syntax. However, the error occurs on random lines in the file, even for commented-out lines. This behavior is not expected and seems to be completely random, affecting various lines throughout the file.

Steps to Reproduce:
Create or edit a custom platform.txt file for a locally created platform (not downloaded via the Boards Manager).
Save the file and attempt to load the platform in the Arduino IDE.
The IDE reports the error "Invalid line format, should be 'key=value'" on random lines, including commented-out lines, which are not supposed to be processed.
Observations:
The error message points to line 1 but the problem persists even after commenting out lines, and it occurs on random lines within the file.
Even valid lines and commented lines trigger errors that shouldn't be flagged.
The issue occurs only with locally created platforms and not with platforms downloaded via the Boards Manager.
The issue is sporadic and random, meaning the error can happen on different lines at different times.
Environment:
Operating System: Linux Mint
Arduino IDE Version: latest of v1
Core: Locally created platform (manually developed, not installed through the Boards Manager).
Additional Notes:
The file is formatted correctly, but the IDE is misreading the lines.
Reinstalling the Arduino IDE or clearing the cache does not resolve the issue.
The problem is isolated to locally created platforms; when the core is uploaded through Boards Manager, there are no errors.

per1234, 2025-01-05:
Thanks for your report @AdamKirby07. Arduino IDE 1.x is no longer under active development. Please try the latest version of Arduino IDE 2.x and then let us know whether the fault still occurs when using that version.

You can download Arduino IDE 2.x from the links listed on the "Software" page of Arduino's website:

https://www.arduino.cc/en/software

AdamKirby07, 2025-01-05:
> Thanks for your report ***@***.***(https://github.com/AdamKirby07). Arduino IDE 1.x is no longer under active development. Please try the latest version of Arduino IDE 2.x and then let us know whether the fault still occurs when using that version.

hi,

The 2.0 dose not recognise local cores.

AdamKirby07, 2025-01-05:
> Thanks for your report [@AdamKirby07](https://github.com/AdamKirby07). Arduino IDE 1.x is no longer under active development. Please try the latest version of Arduino IDE 2.x and then let us know [...]

Version 2.0 is unsuitibel for the development of a core many people have issue s and say to use 1

per1234, 2025-01-05:
> The 2.0 dose not recognise local cores.

It certainly does. I think this will be better handled over on Arduino Forum. Please make a post about the problem over there:

https://forum.arduino.cc/c/development-tools/ide-1-x/18

closed 2025-01-05 by per1234