arduino/Arduino #12028: [IDEA] Allow Other Coding Languages to Be Used for Arduinos

LoganPaxton, 2025-01-22:
So, what if instead of relying solely on C++, Arduino supported languages like JavaScript or Python? According to [Statista](https://www.statista.com/statistics/793628/worldwide-developer-survey-most-used-languages/), in 2024, JavaScript was used by 62.3% of developers, and Python by 51%. Additionally, [SmartBear](https://smartbear.com/blog/best-programming-language-to-learn-first/#:~:text=Python,someone%20who's%20new%20to%20programming.) identifies Python as one of the best languages for beginners. This means Arduino could become an even more accessible platform for newcomers to learn programming, wiring, and hardware design.

By adding support for these widely used languages, Arduino could empower a broader range of people to explore, create, and innovate. While there are third-party options like [Espruino](https://www.espruino.com/) for JavaScript or [MicroPython](https://micropython.org/) for Python, they don’t support every board. For instance, the Arduino Uno R4 WiFi (which I personally use) can’t reliably run MicroPython or Espruino. Ports exist, but they’re often inconsistent or unreliable. Most users prefer official solutions from the original software authors.

If Arduino officially supported JavaScript or Python, it would eliminate the need for third-party firmware and make the platform more beginner-friendly. They could integrate these languages alongside the existing C++ support, not replace it.

Furthermore, while the Arduino Language (a simplified version of C++) is user-friendly, C++ itself is known for its steep learning curve ([Educative.io](https://www.educative.io/blog/is-cpp-hard-to-learn)). Beginners may struggle with it, and the Arduino Language’s limited applicability outside the Arduino ecosystem can be a downside. On the other hand, learning Python or JavaScript offers transferable skills for broader programming opportunities.

So, Arduino, I urge you to consider adding official support for languages like JavaScript or Python. Even Java could be an option. Expanding language support would open the door to new users, new projects, and new ideas.

**TL;DR**:
Arduino currently only offers C++ as the base language for programming. Adding support for languages like JavaScript or Python would make Arduino more accessible, beginner-friendly, and versatile.

LoganPaxton, 2025-01-22:
Sorry, I would post this on the Google group, but I don't have access to the group.

per1234, 2025-01-23:
Hi @LoganPaxton.

>  what if instead of relying solely on C++, Arduino supported languages like JavaScript

That is out of scope for this project. Feel free to start your own project to provide support for programming Arduino boards with JavaScript programs.

>  or Python?

You'll be glad to learn that Arduino has spent years dedicating a significant amount of our finite resources towards providing MicroPython support. Since you are so interested in this subject, I'm surprised you weren't aware of this:

https://blog.arduino.cc/2022/11/10/micropython-officially-becomes-part-of-the-arduino-ecosystem/

closed 2025-01-23 by per1234