arduino/Arduino #12035: [EDIT LABEL REQ!] Rename processing-app-Base to arduino-ide-v1 of 1.18.9

enessmr, 2025-02-25:
Title is.

per1234, 2025-02-26:
Hi @enessmr. It is not clear what you are trying to report or request with this issue. In the future, please make a stronger effort to effectively communicate when you interact with Arduino's repositories.

It is clear that this issue is about something related to the Arduino IDE 1.x codebase. The codebase hosted in this repository is no longer under development, so Arduino would not take action on whatever it is the issue is about. You are welcome to make your own fork of this open source project and modify the code however you like.

closed 2025-02-26 by per1234

enessmr, 2025-03-17:
i see something called processing-app-Base, you might wanna reanme the windowname to arduino-ide-v1